CapnBry/HeaterMeter #15: NEW fan door

marcoemanuelecocco, 2015-06-11:
[image]

I've find a new metod to use the fan without barrel.

please give me a repository to send you the solidworks file and Stl file for 3d print and the open/close configuration for the servo.

[image]
[image]
[image]

marcoemanuelecocco, 2015-06-11:
capsula_1.01.sldprt 
download the object and rename as above
https://cloud.githubusercontent.com/assets/12845292/8107309/ab2046cc-1048-11e5-82f5-0d0578ec82c9.jpg

capsula_1 01 stl
download the object and rename as above
https://cloud.githubusercontent.com/assets/12845292/8107395/21901918-1049-11e5-8d4b-a9d55baa481d.jpg

CapnBry, 2016-01-18:
Looks neat but this is better posted at the user forums
http://tvwbb.com/forumdisplay.php?85-HeaterMeter-DIY-BBQ-Controller

closed 2016-01-18 by CapnBry