CapnBry/HeaterMeter #16: What's the point of the 100k resistors between analog inputs and voltage divider circuits?

bgentry, 2015-09-09:
Hi there,

I apologize in advance for using Issues as a way to ask this question.

I've been learning a lot from HeaterMeter and I'm working on designing my own thermistor sensor board. I noticed that in the v4.2 hardware design, a set of 100k resistors was added between the microcontroller's analog inputs and the standard thermistor voltage dividers.

I've spent a lot of time searching for find some kind of explanation for this kind of circuit design, but all I can find are examples of the simple voltage divider design that was used prior to hardware version 4.2.

Can you explain why these resistors were added and what they do for the circuit?

Thanks!

shmick, 2015-09-09:
The 100k resistors are the R in the RC filter. The 4 x 0.1u caps are the C in the RC filter.

Take a look at the schematic here http://capnbry.net/linkmeter/pcb/hm-4.2/HeaterMeterPi.png

The caps are shown right above IC2 in the schematic.

bgentry, 2015-09-09:
@shmick thanks, I had forgotten about those :)

closed 2015-09-09 by bgentry