CapnBry/HeaterMeter #25: Is there support for Pellet auger

mvogt99, 2015-12-02:
I looked at https://github.com/CapnBry/HeaterMeter/wiki/Blower-and-Servo-Wiring and am wondering if the Servo mentioned was intended to control a Pellet auger? ... or how support for an auger (including into the PID functionality) could be added?

CapnBry, 2015-12-02:
Nope, that's actually something completely different. The pellet auger is either a stepper motor (i.e. the controller sends a single pulse to advance the auger X degrees) or a geared motor (i.e. the motor moves continuously just very slowly due to high gearing). The servo code is for driving a hobby servo, which operates by moving to a position specified by the duration of a continuous pulse. They're fundamentally different and not interchangeable.

I can't think of a way HeaterMeter would control the auger but you might want to [ask on the forums](http://tvwbb.com/forumdisplay.php?85-HeaterMeter-DIY-BBQ-Controller)

closed 2015-12-02 by CapnBry

mvogt99, 2015-12-02:
Thanks for the quick reply... as far as hw goes, I'm guessing something
like this might work... https://www.pololu.com/product/2755 ... just need
to incorporate it into the PID algorythm.  Btw, I believe the auger motor
is a gear motor.

I'll inquire on the forums.

Thanks again,
Mike

Mike Vogt
mvogt99@gmail.com