CapnBry/HeaterMeter #35: wifi config persistent?

Jennnifers, 2016-10-08:
Is there a way to make the wifi settings persistent after boot? 

I followed the instructions here:  https://github.com/CapnBry/HeaterMeter/wiki/HeaterMeter-4.x-Software

I can connect to my wifi by editing the config file every time I boot, but I'm looking for a persistent wifi configuration.

I've been through the "Wireless Client Setup via Web Config step-by-step" multiple times but the config never sticks there either.

CapnBry, 2016-10-08:
The configuration should persist between reboots, This generally indicates that there is a problem writing to the SD card in general, and no information is being saved permanently. You should check under Status -> Kernel Log in the webui and see if you're getting errors at the bottom about "ext4", "extfs", "mmc0", or the like which are causing the filesystem to be "remounted read-only". If you are then you can try re-imaging the SD card to see if it goes away, or trying a different SD card which might be more compatible with the Pi, they are somewhat finicky.

If you need more assistance, I'd recommend creating a thread on [the forums](http://tvwbb.com/forumdisplay.php?85-HeaterMeter-DIY-BBQ-Controller).

closed 2016-10-08 by CapnBry

ChinookTx, 2017-04-22:
I'm having the same problem as @Jennnifers.  I have so far tried 3 micro SD Cards of various sizes, 2 different ways of programming the cards (from linux and MacOS).   Further to this, I have tried a Pi A+ and a Pi Zero.  I cannot get ANY to keep their settings.  This sounds to me like more than a simple glitch with SD Cards??

Jennnifers, 2017-04-23:
I never did solve the problem. 

In fact, now it won't stay connected to wifi for more than a second or two after reconfiguring it.

I've gone through 4 SD cards, 2 usb wifi adapters in the heater meter, and 3 different wifi routers.  Pretty sure it's not a hardware problem at this point.

CapnBry, 2017-04-23:
I can't conceive how this could possibly happen if the configuration changes are applied. Undoing configuration would be a ridiculously complicated challenge if I wanted to do it on purpose because changing the configuration is destructive, there is no backup kept once you hit "Save and Apply". Did you ever look in the kernel log for any indicators that the filesystem was experiencing errors? 

>  You should check under Status -> Kernel Log in the webui and see if you're getting errors at the bottom about "ext4", "extfs", "mmc0", or the like which are causing the filesystem to be "remounted read-only". 

You can also try [downloading a preconfigured snapshot image](https://heatermeter.com/dl/) with your wifi settings baked in, which may work around the issues you're experiencing, but I don't know how I'd even make HeaterMeter do what you're experiencing if I tried.

ChinookTx, 2017-04-23:
I can confirm I wasn't getting any unexpected messages about the sd card in the kernel logs.

I gave a try to the preconfigured image... It worked on the first try on my Pi A+ (which I never got to work previously).  Now I'm really scratching my head.

CapnBry, 2017-04-24:
You and me both! Like I've said, I know a whole bunch about the underlying systems and really can't put together how it is possible to *not* save the network settings across reboots. Great to hear the preconfigured image works though!

ChinookTx, 2017-04-24:
Another odd thing I notice.  The one "normal" image I got to work on my Pi Zero gives me really bad network performance:

> 64 bytes from 192.168.193.3: icmp_seq=27 ttl=64 time=100.149 ms
> 64 bytes from 192.168.193.3: icmp_seq=28 ttl=64 time=20.158 ms
> 64 bytes from 192.168.193.3: icmp_seq=29 ttl=64 time=39.063 ms
> 64 bytes from 192.168.193.3: icmp_seq=30 ttl=64 time=162.846 ms
> 64 bytes from 192.168.193.3: icmp_seq=31 ttl=64 time=79.042 ms
> 64 bytes from 192.168.193.3: icmp_seq=32 ttl=64 time=101.281 ms
> 64 bytes from 192.168.193.3: icmp_seq=33 ttl=64 time=19.417 ms

The one you suggested with burned in settings gives proper ping times:

> 64 bytes from 192.168.193.3: icmp_seq=56 ttl=64 time=4.254 ms
> 64 bytes from 192.168.193.3: icmp_seq=57 ttl=64 time=6.455 ms
> 64 bytes from 192.168.193.3: icmp_seq=58 ttl=64 time=3.663 ms
> 64 bytes from 192.168.193.3: icmp_seq=59 ttl=64 time=4.865 ms
> 64 bytes from 192.168.193.3: icmp_seq=60 ttl=64 time=5.979 ms
> 64 bytes from 192.168.193.3: icmp_seq=61 ttl=64 time=3.630 ms
> 64 bytes from 192.168.193.3: icmp_seq=62 ttl=64 time=3.376 ms

Images appear to be based on different OpenWRT releases, so I suspect something has changed in the firmware that makes it behave better.

Anyway, it works for me now, so unless you want me to try something specific, I'll leave it alone.  More smoking, less tweaking ;-)

CapnBry, 2017-04-30:
I was doing some testing with the old software today and the wifi configuration would reset on every reboot, After some digging I remembered I fixed this issue a while back in the snapshot. What is happening in my case was there was no HeaterMeter board attached so the initialization of the linkmeter package would never complete successfully on boot so OpenWrt would run the initialization again on the next bootup which would reset the wifi back to default. This would also be the case if a HeaterMeter board was attached but not functional, most the network settings would reset each boot until a HeaterMeter was found.

This is fixed in the snapshot firmware, and maybe that's what you folks were seeing? The evidence is that if you ssh into the device, do `ls /etc/uci-defaults/linkmeter` and that file is there, that's it. But if you've moved on there's no reason to look into it any further.

Jennnifers, 2017-04-30:
Thanks CapnBry!  I haven't exactly moved on - I've been using it without any other issues besides the wifi connection.  Would be convenient if I didn't have to get up and go outside to check on it though. :)

I'll take a look and see if there's something wrong with the connection to the heatermeter board when I get a chance.  Wouldn't it just not work without a good connection to the board though? 

If I don't see the linkmeter file, I'll try the preconfigured snapshot image solution instead.

ChinookTx, 2017-05-04:
Thanks for the reply @CapnBry.  I have some spare time today and I'll try a few scenarios.  It might just be that (not having the Heatermeter hooked up).  I initially tried to get the pi going first without the Heatermeter.  Then was a mix of having it on to try and see what IP was reported and off when getting fed up and moving to another Pi.  Rewash, repeat ;-)