CapnBry/HeaterMeter #45: HeaterMeter.com is offline

TheGorf, 2018-09-04:
Can't download any of the required stuff from the heatermeter.com site because it's timing out. Should that be moved to GitHub?  [The instructions from the wiki page](https://github.com/CapnBry/HeaterMeter/wiki/HeaterMeter-4.x-Software) show that I should be going to heatermeter.com/dl to download stuff. But https://downforeveryoneorjustme.com shows the site is offline from multiple locations.  Can we get the downloads from somewhere else?

CapnBry, 2018-09-04:
Hi, thanks for reporting this. I'm not sure why, but the VPS server keeps shutting off. I've got a ticket in to support to figure out why it just shuts off randomly (not shutdown, just like powered off). It started Monday morning.

There's no other way to get the firmware. It's sort of built on the fly by the server and there's a bunch of other light processes that are part of the download and device registration so there's more than just a static file involved. It hasn't been a problem so far as the hosting service has done an amazing job with the uptime, and the volume of HeaterMeter traffic is so low that paying for redundant servers doesn't make much sense.

It has been up for over an hour now, which is a new record for this week so maybe the support people have gotten it straightened out but I haven't heard anything yet.

shmick, 2018-09-04:
Might be good to post the latest LM and HM code to https://github.com/CapnBry/HeaterMeter/releases  ?

CapnBry, 2018-09-04:
I'll consider that for the next release, yeah.

Phortieteaux, 2018-11-17:
Hi Bryan - 

Looks like https://heatermeter.com/ is down again - connection is timing out.  Is there still no path to manually download / configure?

Thanks - 
Jason

CapnBry, 2018-11-17:
I posted about it [on the forums](https://tvwbb.com/showthread.php?75676-HeaterMeter-com-will-be-down-Monday-Oct-8&p=835252&viewfull=1#post835252), but the system is in emergency downtime until tonight.

Phortieteaux, 2018-11-17:
Ah ok, thanks!  I was just able download the firmware directly from the static IP at https://tvwbb.com/showthread.php?75676-HeaterMeter-com-will-be-down-Monday-Oct-8&s=81634f202fd55c300a00c05a2c3550a9, so thought it might be something with DNS.  Thanks for the quick response - and for Heatermeter...  anxious to see if I soldered everything correctly :)

CapnBry, 2018-11-17:
No worries, it looks like it just came back up a few minutes ago but I haven't gotten the all clear that the migration is complete from the host yet. Glad you were able to download the image!

closed 2019-05-20 by CapnBry