CapnBry/HeaterMeter #47: Export CSV metrics

hari-bo, 2019-03-24:
From the HeaterMeter BBQ Control panel you can export the CSV file. 
I cannot find in the documentation how to interpret this values.
The reason I am asking this is that I want to import this data into a graph with time lapse and temperature.

For example:
```
1553424480,nan,14.45652173913,nan,nan,nan,0
1553424540,nan,12.35,nan,nan,nan,0
1553424600,nan,12.236666666667,nan,nan,nan,0
1553424660,nan,12.506666666667,nan,nan,nan,0
1553424720,nan,12.601666666667,nan,nan,nan,0
1553424780,nan,12.511666666667,nan,nan,nan,0
1553424840,nan,12.773333333333,nan,nan,nan,0
```

CapnBry, 2019-03-24:
That is a very valid question! I never realized I had forgotten to add that to the documentation (or even include a header in the CSV file apparently). I'll work on that this week when I get a chance but here's the info:

* UNIX timestamp in UTC
* Setpoint
* Probe 0 temperature (control probe)
* Probe 1 temperature 
* Probe 2 temperature 
* Probe 3 temperature 
* PID output percent (if zero or positive), or lid cooldown remaining (if negative).

All values are the average value recorded over the time period. I see that the first field jumps by 60 each line, so you're looking at 1 minute averages. Note that there are various scales for the data depending on how long the database has been running or what scale you have selected, 10 seconds per period (1 hour of data), 60 seconds (6 hours), 120 seconds (12 hours), or 180 seconds (24 hours).

hari-bo, 2019-03-24:
Thanks! Is there an way to convert the UNIX timestamp to the lead time of the session with the Heatermeter? Because I want to run several sessions with different fuel to compare them with an graph.

CapnBry, 2019-03-25:
Just subtract the timestamp of the first row from all the others. If the data extends all the way through the selected time scale, that won't extend to the beginning of the actual session though since, for example, you're only viewing the most recent hour of data.

HeaterMeter doesn't track the time of session start in any way, so the way I do it for the graphs is to take the full scale CSV (24 hours) and look for the first line that contains an actual number (not a nan). Since we know that each line represents 180 seconds you can tell roughly when the session started from that (now - 24 hours + 180 seconds per line of nans)

CapnBry, 2019-03-26:
The new snapshot firmware includes a header line in the CSV export. The [documentation](https://github.com/CapnBry/HeaterMeter/wiki/Accessing-Raw-Data-Remotely#historical-data) has also been updated with this information. Thanks for bringing it to my attention.

commit 1edaf472 (2019-07-26): [www] Add header to CSV download from home page and archive

Closes #47
closed 2019-03-26 by CapnBry