CapnBry/HeaterMeter #48: compatibility with MQTT server / home assistant

pilot1981, 2019-07-18:
Hi,
it's possible have temperature value in home assistant using MQTT server? 

it could be useful to have all data also when you are outside house...in long time BBQ...

CapnBry, 2019-07-18:
Actually it is weird you'd bring this up, because I just talked with someone else a couple of weeks ago about streaming the data to other systems and I wondered why I didn't already have MQTT built into the HeaterMeter. It is on me todo list now, except it will be a MQTT client, not a server.

pilot1981, 2019-07-19:
thanks for reply.

it's possible to repair link to google photos in wiki section?

CapnBry, 2019-07-19:
It is non-trivial to fix them, since Google Photos doesn't allow linking photos. I've looked at trying to replace them with another image host that allows linking to photos, has albums, and will autogenerate thumbnails in arbitrary resolutions but I couldn't find one easily so figured it wasn't worth the time to stand up and maintain my own image hosting.

The photos are all still there in the new Google Photos albums, but the individual hotlinks for each photo doesn't work.

pilot1981, 2019-07-19:
Ok, thanks.

App HeaterMeter on Apple App Store is yours?

CapnBry, 2019-07-19:
I've actually not heard of that one, but it looks nice. There's also Pit Meter and Pit Droid (for Android) but I'v e not worked on any of these, they are community-created.

kaybee335, 2019-08-20:
I'll add my vote to adding an MQTT client! Now that I have just finished building my unit I was about to write a Python script to scrape that info and publish it to my MQTT broker.

CapnBry, 2019-08-20:
I was hoping to get it in this week but it needs webui to be able to configure it, upload certificates, etc, so it is a bigger project than just getting the package into the build. Also we had that whole deal with migrating to the new webhosting that sort of is still in progress so this is still on my todo list.

kaybee335, 2019-08-21:
I just got mine built yesterday so have no idea about the web hosting issues. Testing last night was amazing. Minimal overshoot after a bit of PID adjustment and stable, stable temps.

I'm in no great rush for the MQTT stuff, I still have to figure out all the stuff that is already there! I'd be happy to give my opinion on things MQTT if you are looking for beta testers.
 
As far as configuration, please allow the user to specify a 'root' of the topic tree that the HM topics would nest under. That would allow easy integration into topic structures that people may already have on their broker, i.e. 'myHouse/automation/outside'.

pilot1981, 2019-11-19:
any progress with MQTT client developing?

CapnBry, 2019-11-19:
Nope, not yet. This ticket will be updated when there's an update on the situation.

CapnBry, 2020-10-13:
The latest snapshot includes a MQTT client in the form of mosquitto_pub and mosquitto_sub. Publishing the streaming status is as easy just piping the output of lmclient into mosquitto_pub:
```
lmclient @LMSS,1 | mosquitto_pub -h myserver -t "heatermeter/hmstatus" -l
```

There's no webui for configuring the server or topic or MQTT login information/certificate, but it should be easy to start working with it with just the line above.

idomp, 2020-10-13:
Amazing work, I would like to test it.
When should I run this command? ssh shell?

and is the mosquitto_pub is this one?
https://manpages.debian.org/testing/mosquitto-clients/mosquitto_pub.1.en.html
inorder to setup user, password and port for the mqtt server.

CapnBry, 2020-10-13:
Yup it is that same mosquitto_pub, which should have TLS support in it too through the existing OpenSSL but I did not test that.

And yeah just SSH in and run that command with the latest snapshot and you should start seeing streaming updates on the MQTT server. If it works and you want it all the time, it can just be added to the rc.local script to start at startup.

gibman, 2020-11-02:
I can confirm it works using mosquito MQTT on home assistant.

I've added this to my startup

lmclient @LMSS,1 | mosquitto_pub -h 192.168.1.174 -p 1883 -u "someuser" -P "somepass" -i "heatermeter" -t "heatermeter/hmstatus" -l

Its a good idea to have the clientid defined here (the 'i' pararmeter)
In my case I also use login/pass

So currently there is no commands I can set from the MQTT consumer side (home assistant)
stuff like : setting alarm thresholds, target temps. etc.

CapnBry, 2020-11-02:
Great, glad it works!

Yeah currently the only way to change anything is through the HTTP API. I'm not sure this will change, at least for a while, because once you start accepting commands you also need authentication, which then needs configuration UI, and so on and becomes a lot of work in addition to creating another API framework which now doubles the amount of work every time new settings are added. Home Assistant can definitely do the HTTP API calls though so you should be able to easily make an MQTT to HTTP gateway if you want to be able to control the HeaterMeter over MQTT.

gibman, 2020-11-02:
About authentication.
Once you have setup the MQTT broker with this in mind you could just blindly accept any command from the MQTT connection as it would be authenticated already :)

I see the problem with doubling the work though.
I guess one could encapsulate the API and expose a subset of MQTT commands. 
This could be done within home assistant etc.

CapnBry, 2020-11-02:
That's a good point, for some reason I was thinking it would have to be a separate connection coming back, but of course one could publish and subscribe on the same outgoing connection. I blame time-change-related morning confusion.

I'm going to leave this ticket open because I'd like to expand on this functionality in the future.

cap9qd, 2020-11-29:
I tried out the mosquitto_pub and it works great from the ssh console!  I can see the data on the RPi that I have the Mosquitto server running on.  However, it doesn't seem to be starting up bootup after adding it to /etc/rc.local.  I found that /etc/rc.local wasn't executable but still didn't work on boot after correcting this item.  Any thoughts?

Also something weird thing is the reboot command doenst work from ssh or from the web gui...?? Maybe something broken in the latest DEV...I used the website to make the "custom" wifi image BTW.

[image]
[image]
[image]

mrand, 2020-11-29:
Here's a discussion about a reboot problem that sounds similar, although the kernel versions don't line up with yours.  Perhaps it was a patch that was applied to multiple kernel versions.  https://www.raspberrypi.org/forums/viewtopic.php?t=264374

As for /etc/rc.local not executing on boot, seems like the most common cause is permissions, which you've already fixed.  Other possible causes are the rc system not actually pointed to the rc.local, and in your case since you're trying to do network access, it's possible that the network isn't up yet.  Here's one of many discussions about rc.local appearing to not work: https://www.raspberrypi.org/forums/viewtopic.php?t=95101

cap9qd, 2020-11-29:
Thanks for the pointers!

I was wondering about networking and may end up making and init script that
depends on network... If I can figure it out. I added a debug message to
print to a file to see if it was executing at all.

I'll do some more digging today and report back.

cap9qd, 2020-11-29:
So I added a debug statement into the rc.local and it is indeed running.  I am almost positive you are right that the wifi isn't up and so it cannot connect to Mosquitto.  I will work on this and report back.

CapnBry, 2020-11-29:
**The Issue system is not for technical support.**

Please discontinue discussion of your problem here and take it to [the support forums](https://tvwbb.com/forums/heatermeter-diy-bbq-controller.85/). rc.local always runs on boot, there is no need to change any permissions. It is likely the reason you can't reboot is because the lmclient command is a BLOCKING command in the startup script.

I'll look at providing assistance tomorrow, but for now please take discussion to the forums.