CapnBry/HeaterMeter #49: "Getting started" guide for hacking on the UI?

marcins, 2019-08-09:
I've been poking around the source of the Web UI and was interested in trying out some changes. Is there a Getting Started guide, or alternatively are you able to provide even some high level guidance as to the easiest way to build & test a new version of the UI?

Is it possible to run the UI locally for development? (even in Docker or a VM?) Is it possible to build a new version of the UI locally and push it to an existing Heatermeter?

marcins, 2019-08-10:
Ah, should've done a bit more digging - looks like I can `ssh` to the system and editing files in `/usr/lib/lua/luci/view/linkmeter` has immediate effect. That's probably enough to get me started.

CapnBry, 2019-08-11:
As you've probably found by now there's not much to the system, just a few lua scripts running a single web page mostly. There's a way to build luci and have it run on x86 but there's so many dependencies that would need to be mocked (the uci config system, network manager, linkmeterd) that there's really not a good reason to do so. To hack away at it, I just use WinSCP and Notepad++ to edit the files and refresh a browser window. All the documentation on the data the backend provides [is in the wiki](https://github.com/CapnBry/HeaterMeter/wiki/Accessing-Raw-Data-Remotely).

If you have further questions, I recommend you join the discussion [on the forums](https://tvwbb.com/forumdisplay.php?85-HeaterMeter-DIY-BBQ-Controller), as this is a bug tracker system and not place to post questions.

closed 2019-08-11 by CapnBry