CapnBry/HeaterMeter #57: 4.0 PCB Compatibility

TristanDoherty, 2020-09-09:
Hi,
I recently upgraded my old and trusty heatermeter (v4.0 PCB with the RCA jack) to the v14 stable software version. Now, my fan does not run while in temperature control mode. 

What doesn't work:
- boot up the heatermeter and connect the pit probe, set a target temp above the current temp
- observe the fan speed indicator (LCD and web UI) is showing 100% fan speed.
- fan does not run, and 0V at the fan connector. I also see 0V at pin 5 of the atmega

What does work:
- boot up the hetermeter and, using the LCD + button, set "Manual Mode" : yes
- using the button, set the fan speed to 100% (all other speeds work as well)
- observe in the web UI that the fan set point is 100%, 12V at the connector and 3.3V at pin 5 of the atmega

Is there something in the v14 software that is no longer backwards compatible with the v4.0 PCB and would be causing this behavior?

TristanDoherty, 2020-09-09:
Just as a followup, I see from the code and the schematics that the 4.0 board is missing the fan voltage feedback circuit that the GrillPid::adjustFeedbackVoltage function uses to adjust the output. 

Is there a way to disable this closed loop output adjustment?

CapnBry, 2020-09-09:
Ah yes, I forgot about it in the forums thread, that if you're upgrading from a software version before the "Voltage" mode was added, you need to go into the webui and change "Fan output mode" to "Pulse" because the older boards do not have the voltage step down/feedback circuitry. This will make the fan run randomly / not at all because it either thinks the fan voltage is already too high or is too low and it can't compensate.

I'd also recommend skipping v14 and going right to the snapshot if you're upgrading. v14 is pretty old and the snapshot has a lot more features, bug fixes, and is just as stable currently.

TristanDoherty, 2020-09-09:
Hi Bryan,

I upgraded to the latest snapshot (SNAPSHOT 20200810 | Load: 0.87 0.55 0.22, AVR Version 20200715B) and made sure that the fan output mode was set to Pulse in the web UI:
Linkmeter->Configuration->PID Output Parameters->Fan Output->Pulse

I am still seeing the same behavior as before though.

I also tried setting the fan output to Voltage and then resetting it to Pulse, but still no effect.

CapnBry, 2020-09-09:
Oh wait a second. Re-reading your initial report, which apologize I didn't fully take in because I was certain it was just a Pulse vs Voltage setting thing, this seems like very strange behavior. I'm going to pull put a 4.0 HeaterMeter for testing but just to confirm, the blower appears to regulate just fine in Manual mode regardless of the Pulse/Voltage output mode setting? It seems to me it should operate randomly in Voltage Mode and perfectly in Pulse mode, but should work the same regardless of Auto vs Manual mode.

TristanDoherty, 2020-09-09:
When I put it in Voltage mode and use the Manual Fan Control, the fan seems to only be full-on or full-off, regardless of the fan % I set (0% is off, 5-100% is max speed). In Pulse mode, I can adjust the fan speed just fine using the Manual Fan Control.

CapnBry, 2020-09-09:
Ok whew, well that's good because it makes sense that it is fully on-off in Voltage mode. I just dug through my box of 100 HeaterMeters and the only 4.0 I have is one that's half assembled, likely a prototype so I'm not sure if I'll be able to get that up and running to test.

Oh! Looking at the settings you've posted, try changing startup max to 100%. That limits max fan speed until temperature is reached so in Auto mode it would be 0% until it reaches temp the first time which sounds like exactly what's happening. It works when you toggle Lid Mode because coming out of lid mode it switches to RECOVERY mode, which also takes it out of STARTUP mode.

TristanDoherty, 2020-09-09:
Yes! That was totally it. I changed the Fan Startup Max and things are back to normal.
Thank you!
I'll close the issue (especially considering it wasn't really an issue, just a user who didn't know the right setting!)

closed 2020-09-09 by TristanDoherty

CapnBry, 2020-09-09:
Glad we got it worked out and happy to help an ole 4.0 user. I was certain it should still work with the latest software and it is good news to hear it does!