CapnBry/HeaterMeter #59: Port for ESP32

Bananamannn, 2021-09-05:
Hi All,

I'm not much of a programmer but how much work would be involved in porting this from a Raspberry Pi / AT Mega unit too an ESP32 based system?

CapnBry, 2021-09-06:
Oh you'd just have to write it all over again, everything except the home page javascript/html (which would also need some changes). One main drawback of the ESP platform is storage. The database is over 100KB so it would take a significant amount of RAM to hold it, or some new scheme would have to be devised to store parts in flash as well without writing to flash too much to wear it out. Even a Raspberry Pi Zero can also blow the doors off any ESP when it comes to serving webpages so you'd do an entire rewrite to end up with a less capable device for a couple dollars lower cost.

Bananamannn, 2021-09-07:
Oh wow okay. That sounds like a bit of an exercise, sorry for bothering you with it.

I might take your advice and try and find myself an older Pi to use.

closed 2021-09-07 by Bananamannn