CapnBry/HeaterMeter #61: Upgrading AVR firmware

brafster, 2022-08-03:
Hi,

quite new to this... I bought a pre-compiled 4.2 board online. Currently it's running AVR 20140630B. Can I just install the latest version from the online repository, or does it also show incompatible versions for 4.3 for instance? Just want to make sure I don't break anything :-) The latest one listed is snapshots/trunk/heatermeter.hex

[image]

CapnBry, 2022-08-03:
All the 4.x hardware is compatible with the latest firmware. However, I highly recommend you upgrade the Pi firmware image before upgrading the AVR firmware, as AVR firmware assumes the Pi firmware is up to date.
https://heatermeter.com/dl/

Just use the `download .gz` link and upload it from the System -> Backup / Update Firmware page. Your wifi config should be preserved so long as the Keep Settings checkbox is checked.
[image]

brafster, 2022-08-04:
Awesome! Thanks so much for the help.

closed 2022-08-04 by brafster