CapnBry/HeaterMeter #63: New wireless thermometer option

rjroch, 2023-05-07:
I've been using a HeaterMeter 4.2.something for like 8 years, and it has been awesome. I've been reading recently that you were trying to incorporate wireless temperature sensors into HeaterMeter, including using Meater probes (which apparently didn't pan out).

Just wanted to drop a note that Chris Young of Modernist Cuisine and Joule fame has released a thermometer that might fit the bill, the Combustion Predictive Thermometer. There's a great developer page at https://developer.combustion.inc/, but it's got BLE connection, and there's no pairing needed, so the thermometers essentially spray temp data from 8 temp sensors out into the world. Shows core temp, surface cook temp, ambient temp, and works up to like 500-600 degrees. I ordered a set of them, and they seem incredibly good so far.

Hoping you might someday be in a position to take a crack at integrating these!

CapnBry, 2023-05-08:
Hello! I haven't looked at wireless thermometer integration in many years. I was hesitant to do any work at all with the Meater due to it possibly just ending up a failed product that would disappear after going through all the development work to hack it in, or it being an ongoing support problem as their protocol kept changing over time. I'm shocked to see it is still exists actually, but it does make me pleased that they were able to create a successful product.

I doubt there will be any integration of any wireless thermometers into HeaterMeter though, there's just too much work needed for a project that's on its way out. The wireless grill thermometer market has a hundred different options now at price points that beat HeaterMeters and connectivity options that beat HeaterMeter's 14 year old design.

There's a lot of challenges that would need to be addressed:
* Bring the firmware over from LEDE to something new, so up to date bluetooth tools are available
* Integrate bluetooth into the system: services, webui, configuration, binding incoming data to HeaterMeter probes, etc
* Allow the HeaterMeter software to accept external data both into the web services, main daemon, and publishing them into the core controller (which comes with a large set of arbitration challenges as multiple services try to talk to the MCU)

It's just a ton of work to integrate and there's just not a lot of life left in the project with existing hardware. It would certainly be a better idea to just start from scratch and rewrite everything to use more modern hardware and not use Raspberry Pis due to their extreme cost now.

rjroch, 2023-05-08:
Man, thanks for replying! I had considered a bit of the work that would need to be done, like the need to integrate bluetooth and how to bind data, but that's a long list. I didn't even consider the Raspberry Pi struggles at all since I have a box of various versions more or less laying around from the good old days. I guess I was hoping you had a framework ready to go, just waiting for data to be dropped in.

It does seem like it would make sense to start from scratch. That said, as much as I like everything being integrated together in one place, the more I think about it, a wireless-only controller device could probably be done as an all-in-one ESP32-based fan+servo damper. With all these new devices having their own apps, it doesn't make sense to reinvent those pieces.

In any case, there's no reason to bug you further. Thanks again for all the hard work you did over the years. What a great device the HeaterMeter has been.

closed 2023-05-08 by rjroch

CapnBry, 2023-05-08:
hehe yeah the Raspberry Pi struggle is real, it was sort of the last nail in the coffin for the project after going through all the other supply chain in the past few years. It is a bit of a tough sell to have a "cheap grill controller" project when the Pi part is going for $60-$100 shipped as just one component. I have a handful I shuffle between different tasks, but for anyone new who wants in, I have a hard time recommending them for the price, especially considering it has to be dedicated to being a HeaterMeter.

Creating a tiny firmware image based on OpenWrt was great for making it quick to install and boot super fast but really makes it tough to extend with things like bluetooth support or for anyone to be able to extend really. I don't know which it would be harder to do, move the project from LEDE back onto OpenWrt (including fixing all the tiny bugs that would return) then hope there's a decent bluetooth OpenWrt system, or convert the entirety of the LuCI webui  / controller / daemon code to something else and use Raspberry Pi OS. Either is a lot of work before even starting on figuring out how to manage attaching to other devices and creating the infrastructure to pass it to linkmeter which then needs to pass it to HeaterMeter.

There weren't even any wifi-enabled MCUs when the project started, and the first HeaterMeter used an external $60 wifi module over serial. The ESP8266 came along later but couldn't do everything the project needed, but now there are amazing ESP32s like the S3 that have native USB support for easy flashing or config, dual cores, tons of RAM, 4MB of storage and plenty of I/O. You could almost do everything right from one chip, although storing the current running database might be a bit of a problem, and their ADCs aren't all that good. In any case, I just feel like making an embedded PID controller isn't the complicated part any more and the HeaterMeter architecture is too complicated to build on considering the other options.

I'm happy to hear you still love your HeaterMeter though, I still use mine all the time and it has become so trusted I sleep all night with it running. It was a great project and I hope it lasts you many years to come.

theschles, 2023-08-26:
Hi @CapnBry I understand where you're coming from.  

I built my Heatermeter in 2014 or 2015 with a Raspberry Pi 1B.  It was my second through-hole soldering project.  

It's been my go-to for barbecuing and smoking meats ever since.

If you or @rjroch or anybody does want to start up a new project -- perhaps mixing in Home Assistant for interoperability -- please let me know?