CapnBry/HeaterMeter #65: Is the project dead?

asdkjasdfla, 2024-05-28:
There haven't been any updates on this project for a while. The website for heatermeter is also down. Is this project dead?

CapnBry, 2024-06-10:
The HeaterMeter website is still fine, but the store is closed. I'm not really working on any new updates for the software either but I will keep it running as I still use my HeaterMeters all the time.

asdkjasdfla, 2024-06-10:
Thankyou. Any idea if those who haven't got the hardware for heatermeter can get it from?

CapnBry, 2024-06-10:
[The parts list is in the wiki](https://github.com/CapnBry/HeaterMeter/wiki/HeaterMeter-4.3-Hardware), so I'd suggest an online electronics retailer, such as Mouser. The PCBs can be made even cheaper than when I was making them by using PCBWay or similar, but you'll have to buy at least 5 I think so it would be a good idea to try to do some sort of group buy.